openmv/openmv #412: nn_quantizer.py runs indefinitely

laoonatic, 2018-11-16:
Hi,

I am trying to quantize a modified model based on the lenet trained on the MNIST data set. I only retrained the last layer from the lenet_train_test.prototxt in [1] on my own data set. So what I did was:
- Change the input of the data layer to have the location on my new data
- Change the learning rates of all the layers except for the last one to 0
- Load the weights of the lenet model trained on the MNIST
- Train only the last layer

It all works well, except when I try to quantize the model using `nn_quantizer.py`. It doesn't seem to print any errors, but it keeps on hanging in a loop and run indefinitely. The output is the following:
```
$ python2 nn_quantizer.py --model data/lenet_train_test_modified.prototxt --weights data/weights.pretrained.caffemodel --save data/lenet_modified.pkl
WARNING: Logging before InitGoogleLogging() is written to STDERR
W1116 10:39:16.540519 16351 _caffe.cpp:139] DEPRECATION WARNING - deprecated use of Python interface
W1116 10:39:16.540547 16351 _caffe.cpp:140] Use this instead (with the named "weights" parameter):
W1116 10:39:16.540557 16351 _caffe.cpp:142] Net('data/lenet_train_test_modified.prototxt', 1, weights='data/weights.pretrained.caffemodel')
I1116 10:39:16.541888 16351 net.cpp:294] The NetState phase (1) differed from the phase (0) specified by a rule in layer mnist
I1116 10:39:16.541908 16351 net.cpp:51] Initializing net from parameters: 
name: "lenet_modified"
state {
  phase: TEST
  level: 0
}
layer {
  name: "mnist"
  type: "HDF5Data"
  top: "data"
  top: "label"
  include {
    phase: TEST
  }
  hdf5_data_param {
    source: "test_h5_list.txt"
    batch_size: 50
  }
}
layer {
  name: "conv1"
  type: "Convolution"
  bottom: "data"
  top: "conv1"
  param {
    lr_mult: 0
  }
  param {
    lr_mult: 0
  }
  convolution_param {
    num_output: 20
    kernel_size: 5
    stride: 1
    weight_filler {
      type: "xavier"
    }
    bias_filler {
      type: "constant"
    }
  }
}
layer {
  name: "pool1"
  type: "Pooling"
  bottom: "conv1"
  top: "pool1"
  pooling_param {
    pool: MAX
    kernel_size: 2
    stride: 2
  }
}
layer {
[... 200 lines cut ...]
I1116 10:39:16.547703 16351 net.cpp:200] pool2 does not need backward computation.
I1116 10:39:16.547708 16351 net.cpp:200] conv2 does not need backward computation.
I1116 10:39:16.547713 16351 net.cpp:200] pool1 does not need backward computation.
I1116 10:39:16.547719 16351 net.cpp:200] conv1 does not need backward computation.
I1116 10:39:16.547724 16351 net.cpp:200] label_mnist_1_split does not need backward computation.
I1116 10:39:16.547730 16351 net.cpp:200] mnist does not need backward computation.
I1116 10:39:16.547735 16351 net.cpp:242] This network produces output accuracy
I1116 10:39:16.547740 16351 net.cpp:242] This network produces output loss
I1116 10:39:16.547751 16351 net.cpp:255] Network initialization done.
```

When I force stop the script, it seems to be hanging in the following place:
```
^CTraceback (most recent call last):
  File "nn_quantizer.py", line 608, in <module>
    my_model.get_graph_connectivity()
  File "nn_quantizer.py", line 241, in get_graph_connectivity
    if (current_blob in value) and (key not in traversed) and \
KeyboardInterrupt
```
I tried using pdb to identify the problem, but I still don't seem to get what's wrong, especially as I don't get any error message. Also, the network uses the same types of layers as in the MNIST model example, so I expected it to work here. The only difference is the data layer, which takes as input an HDF5 database (as opposed to an LMDB one). However, I didn't find any warning about this not being supported.

If anybody has any ideas about what's going wrong, I would be grateful. Thank you!

[1] https://github.com/BVLC/caffe/tree/master/examples/mnist

laoonatic, 2018-11-16:
An update: I tried running the quantizer on the same model, but using an LMDB data layer for the input and it worked. So the issue is probably the HDF5 data layer. Is this issue known? If yes, then maybe consider adding a warning or something about it in the documentation?

kwagyeman, 2018-11-16:
Hi, I still need to merge this PR: https://github.com/kwagyeman/openmv/tree/more_nn_nets/ml/cmsisnn

kwagyeman, 2018-11-16:
It is a lot more documented.

iabdalkader, 2018-11-16:
Those are the supported layers:
https://github.com/openmv/openmv/blob/master/ml/cmsisnn/nn_quantizer.py#L219

The type needs to be "data" or "5" for the data layer:
https://github.com/openmv/openmv/blob/master/ml/cmsisnn/nn_quantizer.py#L201

Nout our converter script expects "data" for the data layer type.

laoonatic, 2018-11-19:
Thank you for your answers, they are really helpful.

closed 2018-11-21 by iabdalkader