openmv/openmv #414: No support for loading jpg object into RAM

raryanpur, 2018-12-01:
### Steps to reproduce

Try to load a jpg image with the `image.Image` constructor, as in

```
img = image.Image("/image.jpg", copy_to_fb=True)
```

### Expected behaviour

Loading a jpg into RAM using the `Image` class constructor shouldn't throw an error. Per the documentation for `class image.Image`, the `image.Image(path[, copy_to_fb=False])` constructor supports both `jpg` and `jpeg` image files. At the least, there should be support to load a `jpg` object into RAM.

### Actual behaviour
Loading a `jpg` image into RAM using the `Image` class constructor throws an error `OSError: Unsupported format!`

iabdalkader, 2018-12-04:
Actually the library doesn't support loading JPEG image. This needs to be fixed in the docs @kwagyeman.

kwagyeman, 2018-12-04:
Can you add this feature @ibrahim.

kwagyeman, 2018-12-04:
@iabdalkader

raryanpur, 2018-12-05:
Thanks guys!

iabdalkader, 2020-12-01:
Fixed, many years ago :sweat_smile:

closed 2020-12-01 by iabdalkader