openmv/openmv #415: Subpixel corners for apriltag.corners()

edmc2, 2018-12-09:
### Steps to reproduce
1.Run find_apriltags() on an image
2.Run corners() on the returned tag object/objects
3.

### Expected behaviour
Tell us what should happen
corner positions should be returned to hundredths of a pixel precision

### Actual behaviour
Tell us what happens instead
corner positions are returned to pixel (integer) precision

### Describe your setup

**IDE version:**
2.0.0
**Firmware version:**
3.0.0
**Operating system:**
Windows 10
### For developers

**GCC version:**

**Build target:**

iabdalkader, 2018-12-10:
@kwagyeman

kwagyeman, 2023-11-22:
This was merged in. The current apriltag library doesn't output subpixel positions for the corners.

closed 2023-11-22 by kwagyeman