openmv/openmv #416: How can I find minimum area rectangle in binary image?

midnightpen, 2018-12-10:
I have EmguCV project to find all minimum area rectangle in contour like below.

VB.NET with EmguCV code :
imgRects.Draw(CvInvoke.MinAreaRect(contour), New Bgr(Color.Red), 3)

I use visual studio with emgu cv to measures width and length of object. In last project, I use openmv with non-distortion lens. Then I want to make embeded machine to use mesure width and length of object and display when found defect. How can I find minimum area rectangle in binary image like vb.net code as above?

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

iabdalkader, 2018-12-10:
Hi, please use the forums to post your questions.

kwagyeman, 2018-12-10:
I asked him to post this here.

kwagyeman, 2019-02-04:
Hi, this has been added.

closed 2019-02-04 by kwagyeman