openmv/openmv #430: regarding mt9v034 working mode.

zx2624, 2019-01-22:
I've been reading mt9v034.c, yet I can't seem to find the code about changing working mode. I'm doing this because I want to activate the snapshot mode of mt9v034 for synchronization between this camera and a IMU(mpu6050). Am I missing something here?

kwagyeman, 2019-01-22:
Um, there's no code in there yet for enabling the trigger. Did you get one of the pre-release H7 boards? You can enable the trigger by doing a register write to a particular register address. See the datasheet.

The driver state for the lepton and the mt sensor is not fully complete. The drivers will be completed before the product goes on sale.

zx2624, 2019-01-22:
Thanks a lot !
No I didn't get the pre-release board, I'm just looking for some inspiration from your project since I know little about hardware. Thank you for your suggestion.

closed 2019-01-22 by zx2624