openmv/openmv #444: Face detection example program find_features() throws exception on sample H7

2a25ARBBz8S35kynSxwAMJc7, 2019-02-06:
### Steps to reproduce
1. Install OpenMV ide 2.1.0
2. Connect sample H7 camera and update firmware to 3.2.0 
3. Execute example program face_detection_1.py 

### Expected behaviour
Example program should run without error. Program does run without error when M7 camera is connected instead of H7.

### Actual behaviour
OpenMV ide throws exception: "OSError: Invalid ROI dimensions!" This function call should not require ROI dimensions.

### Describe your setup
**IDE version:**
2.1.0
**Firmware version:**
3.2.0
**Operating system:**
Windows 7 on Lenovo T440p
### For developers

**GCC version:**

**Build target:**

iabdalkader, 2019-02-06:
Hi, I tested the face_detection.py example with FW 3.2.0 and it's running with no exceptions. Did you change the script ?

kwagyeman, 2019-02-06:
I already sent him the H7 firmware. There should be no issue.

2a25ARBBz8S35kynSxwAMJc7, 2019-02-06:
Ibrahim,

Thanks for looking at this issue. The exception was thrown by the unaltered face_detection_1.py example program on H7 FW 3.2.0 in IDE 2.1.0. I am attaching a snip of the screen. I am still learning OpenMV so perhaps I am doing something dumb without realizing it.

Best regards,

Steve Welch

kwagyeman, 2019-02-15:
This apparently is an issue on the global shutter camera only. Will need to check.

iabdalkader, 2019-03-01:
This issue has been fixed.

closed 2019-03-01 by iabdalkader