openmv/openmv #456: how to frozen py modules?

HarryZeng, 2019-02-22:
### Steps to reproduce
1. make FROZEN_MPY_DIR=modules，display ERROR:undefined reference to `fb_alloc_free_till_mark'
and then I Comment “fb_alloc_fb_alloc_free_till_mark” in nlr.h.after do that ,is works
2.And  "make FROZEN_MPY_DIR=modules "  again，Error........
3.why?How can I frozen .py modules in firmware??

ERROR  LOG:
----
make[1]: Leaving directory '/home/harry/openmv/openmv/src/uvc'
/home/harry/openmv/openmv/src/build/micropython/py/builtinhelp.o: In function `mp_help_print_obj':
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/builtinhelp.c:154: undefined reference to `mp_frozen_mpy_names'
/home/harry/openmv/openmv/src/build/micropython/py/frozenmod.o: In function `mp_frozen_stat':
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/frozenmod.c:136: undefined reference to `mp_frozen_mpy_names'
/home/harry/openmv/openmv/src/build/micropython/py/frozenmod.o: In function `mp_find_frozen_module':
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/frozenmod.c:153: undefined reference to `mp_frozen_mpy_names'
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/frozenmod.c:153: undefined reference to `mp_frozen_mpy_content'
/home/harry/openmv/openmv/src/build/micropython/py/qstr.o: In function `mp_thread_mutex_init':
/home/harry/openmv/openmv/src/micropython/ports/stm32/./mpthreadport.h:44: undefined reference to `mp_qstr_frozen_const_pool'
/home/harry/openmv/openmv/src/build/micropython/py/qstr.o: In function `qstr_pool_info':
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/qstr.c:284: undefined reference to `mp_qstr_frozen_const_pool'
/home/harry/openmv/openmv/src/build/micropython/py/qstr.o: In function `qstr_dump_data':
/home/harry/openmv/openmv/src/micropython/ports/stm32/../../py/qstr.c:295: undefined reference to `mp_qstr_frozen_const_pool'
collect2: error: ld returned 1 exit status
Makefile:498: recipe for target 'firmware' failed
make: *** [firmware] Error 1
--------
### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2019-03-11:
Hi, I don't know the error here. However, we haven't been maintaining frozen modules.

iabdalkader, 2019-03-13:
It's not okay to just comment `fb_alloc_fb_alloc_free_till_mark` if you build the firmware without this function, fb memory will not be free'd. Maybe just comment it so you can build mpy-cross (still not sure if this is okay).

The other errors may be related to this issue:
https://github.com/micropython/micropython/issues/4333

ddech, 2019-04-27:
i solve this problem by adding dummy file "openmv/src/micropython/py/omv_dummy.c"
in that dummy file contain a dummy `fb_alloc_fb_alloc_free_till_mark` function with the weak attribute.
```
void __attribute__((weak)) fb_alloc_free_till_mark()
{
}
```
then, in "openmv/src/micropython/py/py.mk" append `omv_dummy.o` to the `PY_CORE_O_BASENAME` list.

finally, in "openmv/src/Makefile" after MicroPy Objects list (about line 315) add this
```
ifneq ($(FROZEN_MPY_DIR),)
FIRM_OBJ += $(addprefix $(BUILD)/$(MICROPY_DIR)/,frozen_mpy.o)
endif
```

when micropython is compiling, a dummy `fb_alloc_free_till_mark` will be linked. 
but in the final link, the real `fb_alloc_free_till_mark` will replace a dummy.

and an error undefined reference to `mp_frozen_mpy_names', `mp_frozen_mpy_names', `mp_frozen_mpy_content' and `mp_qstr_frozen_const_pool', the cause is `frozen_mpy.o` object is not included in project.

kwagyeman, 2019-04-28:
Hi, @ddech - thanks for the previous PR. Please send one for this too! :)

iabdalkader, 2019-04-30:
I think this is now fixed.

closed 2019-04-30 by iabdalkader