openmv/openmv #460: `RuntimeError: maximum recursion depth exceeded` on OpenMV3 with timer

kidswong999, 2019-02-26:
```
import sensor, image, time
from pyb import Pin, Timer
import micropython
micropython.alloc_emergency_exception_buf(100)

sensor.reset()                      # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)     # Wait for settings take effect.
clock = time.clock()                # Create a clock object to track the FPS.

def systick(timer):            # we will receive the timer object when being called
    print("yes")
tim = Timer(2, freq=10)
tim.callback(systick)

while(True):
    clock.tick()                    # Update the FPS clock.
    img = sensor.snapshot()         # Take a picture and return the image.
    print(clock.fps())              # Note: OpenMV Cam runs about half as fast when connected
                                    # to the IDE. The FPS should increase once disconnected.
```

RuntimeError: maximum recursion depth exceeded
firmware: 3.2.0 OpenMV3

by the way, OpenMV4 did not appear error.

kwagyeman, 2019-02-26:
@iabdalkader - I can't see what's wrong. Can you debug?

zyafly, 2019-02-27:
I tried it and found that after defining a timer and then adding this code(img = sensor.snapshot()), the following error will occur.

zyafly, 2019-02-27:
RuntimeError: maximum recursion depth exceeded.firmware: 3.3.0 OpenMV3

zyafly, 2019-02-27:
> @iabdalkader - 我看不出有什么问题。你可以调试吗？
I tried it and found that after defining a timer and then adding this code(img = sensor.snapshot()), the following error will **occur:RuntimeError: maximum recursion depth exceeded.**firmware: 3.3.0 OpenMV3

iabdalkader, 2019-02-27:
@kidswong999 You can't call any functions that allocate memory in callback functions. Try toggling an LED instead:

```
import pyb
led = pyb.LED(1)
def systick(timer):            # we will receive the timer object when being called
    led.toggle()

```

closed 2019-02-27 by iabdalkader