openmv/openmv #475: OSError: failed to init WINC1500 module

BebeSparkelSparkel, 2019-03-09:
### Steps to reproduce
1. Plug in Wifi Shield to OpenMV Cam
2. Plug into computer with micro-usb cabel
3. Erase drive and rewrite files
4. Connect to device
5. Run connect.py

### Expected behaviour
Expect to establish a wifi connection

### Actual behaviour
Received pop up of `OSError: failed to init WINC1500 module`

### Describe your setup

**Hardware:**
- Wifi Shield R2 ATWINC1500-MR210PB 2/11/2016
- OpenMV Cam M7 - OV7725

**IDE version:**
2.1.0

**Firmware version:**
3.2.0

**Operating system:**
macOS Mojave 10.14.3 (18D109)

kwagyeman, 2019-03-10:
Hi, this would be an issue with the hardware and not the software. Please try to debug the issue on the forums and if not possible we can ship another wifi shield. In particular, please verify there's a slid electrical connection between all the pins.

closed 2019-03-10 by kwagyeman

kwagyeman, 2019-03-12:
> Hello Kwab,
>
> I'm the shield owner. We have been trying to make it work for months to no avail. About 4-5 of us have tried with no luck. The connections seem to be okay, but I don't have a microscope. If you don't mind, could we try replacing it please?
>
> Kind regards,
>
> -Alex

Yeah, we can do that. Um, can you place an order for another shield on our website and then I can refund the order after it ships. We have an automated system that makes this transaction path very easy.

-- Kwabena

Sashulik, 2019-03-13:
Done. Order #3659.