openmv/openmv #491: Firmware 3.3 unable to read file from SD-card

thyrrestrup, 2019-03-20:
### Steps to reproduce
1. FAT32 format sd-card
2. Update to firmware version 3.3 on H7
3. load cifar10.network on sd-card
4. run cifar10 example from OpenMV IDE

If I downgrading to 3.2 without making any changes to sd-card then everything runs as expected.

### Expected behaviour
Example runs

### Actual behaviour
OSError: The file/directory object is invalid

### Describe your setup

**IDE version:**
2.1.0

**Firmware version:**
3.3/H7

**Operating system:**
Mac OS X/Ubuntu 18.04

kwagyeman, 2019-03-20:
Hi, Ibrahim worked on the file system code given some issues saving mjpeg files. Um, does the path to your network have a leading forward slash on it?

thyrrestrup, 2019-03-20:
Yes it does. Should I try without?

kwagyeman, 2019-03-20:
Hmm, okay, can you email openmv@openmv.io with the script and the net? Ibrahim will test it out and then fix whatever is bugged and re-release.

thyrrestrup, 2019-03-20:
Yes I can do that tomorrow from work - but it’s just the default cifar10.network file and default cifar10 example script.

iabdalkader, 2019-03-21:
There's no need for that, I found the bug and I'm fixing it, we'll release 3.3.1 soon.

closed 2019-03-21 by iabdalkader

thyrrestrup, 2019-03-21:
No problem! Thanks