openmv/openmv #500: When to add support for native code generation (native, viper, inline asm) to the openmv?

kingsjl, 2019-04-02:
When to add support for native code generation (native, viper, inline asm) to the openmv?

kwagyeman, 2019-04-03:
Um, it can probably be enabled if done so in the config port header.

@umux - Can you turn it on?

iabdalkader, 2019-04-03:
I'm not sure which option you want enabled, almost everything is enabled. If you know which one let us know and we'll enable it.

iabdalkader, 2019-09-09:
Hey, I checked this and all native emitters are enabled but they only work on the F7 (OpenMV-3). For some reason (related to cache I think) the H7 crashes when running native code, I'll into this later. Here's an example that works on the F7:

```
@micropython.native
def native(a, b):
    print(a + b)

@micropython.viper
def viper(a, b):
    return a + b

@micropython.asm_thumb
def asm():
    movw(r0, 42)

print(asm())
print(viper(1, 2))
native(1, 2)
```

iabdalkader, 2019-10-28:
Native code emitters are working now on the H7.

closed 2019-10-28 by iabdalkader