openmv/openmv #507: Files for lenet ml

bryansosa, 2019-04-18:
Hi there,

I was wondering if you can provide the files:

caffe/examples/mnist/mean.binaryproto
models/lenet/lenet_iter_10000.caffemodel

I am running an experiment on a m3 processor but I am having a hard time to generate those files.

I not possible to upload to github, can you send them to me personally? I would appreciate that!

Thank you very much beforehand.

kwagyeman, 2019-04-18:
Hi, we provide the converted model output. What's the need of the iter model?

bryansosa, 2019-04-18:
I want to generate the c code as in https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10

kwagyeman, 2019-04-18:
I see. Um, I have the left over files on my deep learning desktop. I can get them to you later.

bryansosa, 2019-04-18:
I would really appreciate this! Thanks a lot :) ... Will you upload them to the repository or send them to me personally?

kwagyeman, 2019-04-18:
Um, well, I could send you the files in a zip. I think you need more than just the itermodel.

bryansosa, 2019-04-18:
Ok you can send them to bsosabar@uccs.edu 

Thanks!
Bryan.

bryansosa, 2019-04-19:
Hi @kwagyeman ,

I wonder if you were able to get a hold on the ML file?

Thank you very much.

kwagyeman, 2019-04-20:
Sent.

closed 2019-04-20 by kwagyeman