openmv/openmv #513: binary function wrong in RGB565

kidswong999, 2019-04-25:
example code:
https://github.com/openmv/openmv/blob/master/scripts/examples/04-Image-Filters/erode_and_dilate.py

when Grayscale, it works great.
[image]

when RGB565, the white turned gray.
[image]

Maybe there are bugs in binary function.https://github.com/openmv/openmv/blob/1412a2d23410436091be84ff3fc6b5b520cf8061/src/omv/img/binary.c

**IDE version:** 2.2.0
**Firmware version:** 3.3.1

ychxing, 2019-04-26:
Function binary() doesn't work correctly on my OpenMV3 board after updated firmware version to 3.3.1

kwagyeman, 2019-04-26:
Will fix.

kwagyeman, 2019-04-29:
Fixed, was due to some missing parens on a define.

closed 2019-04-29 by kwagyeman

ychxing, 2019-04-29:
when will upgrade a new fixed firmware?