openmv/openmv #521: Logic error in DMA 4 bytes alignment check.

Lak4CYUT, 2019-05-02:
OpenMV3 R2 (STM32F7XX)

### Expected behaviour
When target address is aligned on 4 bytes, read/write SPI should use the DMA.

### Actual behaviour
Not work.

### Describe your setup
I found this issue from the source code.

In micropython/ports/stm32/spi.c defined below macro.

`#if defined(STM32F7)`
`#define DMA_BUFFER(p)       ((uint32_t)p & 3)`

It means, if the address is **NOT** aligned on 4 bytes, it will return non-zero, a true.
Then rely on this macro to decide whether to use DMA to read/write SPI.
```
if (len == 1 || query_irq() == IRQ_STATE_DISABLED || !DMA_BUFFER(src)) {
...
status = HAL_SPI_Transmit(self->spi, (uint8_t*)src, len, timeout);
...
} else {
...
status = HAL_SPI_Transmit_DMA(self->spi, (uint8_t*)src, l);
...
```
Please focus on the if statement, when the address is **aligned on 4 bytes**, DMA_BUFFER(src) return false (0x00), through a ! (not), it become true.
**So, it will going to execute HAL_SPI_Transmit(), not HAL_SPI_Transmit_DMA().**

HAL_SPI_Transmit_DMA() will be executed only when the address is **NOT** aligned on 4 bytes.
**It should be a logic bug.**
ex: (not only)
[https://github.com/openmv/micropython/blob/openmv/ports/stm32/spi.c#L457](url)

**And MicroPython master branch is no this bug.**

iabdalkader, 2019-05-02:
This has been fixed before in sdcard.c but I forgot to include this fix in spi.c. I will fix it now, thanks for reporting this bug.

iabdalkader, 2019-05-02:
Fixed in #522.

closed 2019-05-02 by iabdalkader