openmv/openmv #524: FileSystem corruption during power loss

Vinschni, 2019-05-04:
Is there any way to avoid file system corruption if openmv loses power during recording with software meachnisms?

Both filesystems available on the openmv suffer from corruption if power loss occurs during SD access. Whole folders or even whole file system can be corrupted.

Fat32 and exFAT are both are not failesafe.

OpenMV does not integrate a power-loss resiliant FS like LittleFS currently. Anything I can do in software to avoid corruption of whole folders on power loss? That the accessed file is corrupted is legitimate.

### Steps to reproduce
1. Record 10s chunks of video to SD card, each in a own file
2. Switch off power
3. Switch on power

### Expected behaviour
The last file, during which the power loss occured is corrupted or not existing.

### Actual behaviour
File system of SD card is corrupted. If files written to folder whole folder is corrupted, no file can be read in folder.

### Describe your setup

**IDE version:** 2.2.0

**Firmware version:** 3.3.1

**Operating system:** Windows 10

kwagyeman, 2019-05-04:
Yeah, this is a legit problem. Given we have to use FAT so operating systems can mount the drive I don't really know what a solution for you would be. I'd upstream this ask to MicroPython. We'll follow whatever lead they implement.

kwagyeman, 2019-05-04:
Um, regarding handling this type of stuff in software. Well, I'd add a lot of extra capacitance to the board and have an interrupt line setup for detecting power loss and then shutting the system down on this.

kwagyeman, 2020-05-13:
littlefs is now in MicroPython v1.12.

It's not easy to use... but, it's in the firmware now.

http://docs.openmv.io/reference/filesystem.html#hybrid-stm32

Let us know if the feature complains about not being enabled and we can enable it.

You should format the SD card. That said... this stuff is just really low level. I can't say it's that easy to use right now.

Anyway, we will support pulling all features of this into our firmware.

closed 2020-05-13 by kwagyeman