openmv/openmv #527: OSError: failed to init WINC1500 module (H7)

tpag26, 2019-05-11:
**Steps to reproduce**
Plug in Wifi Shield to OpenMV Cam
Plug into computer with micro-usb cable
Update firmware
Connect to device
Run scan_1.py

**Expected behaviour**
Print scan results

**Actual behaviour**
Received pop up of OSError: failed to init WINC1500 module. Have tried two different cameras to rule out a hardware issue with camera.

**Hardware:**
Wifi Shield R2 ATWINC1500-MR210PB 2/11/2016
OpenMV Cam H7 - R1

**IDE version:**
2.2.0

**Firmware version:**
3.3.1

**Operating system:**
Win10 x64

iabdalkader, 2019-05-11:
Hi, I just tested mine with 3.3.1 and it's working, it's not a firmware issue. Please make sure all the pins are connected, if it still doesn't work it's probably a bad WiFi shield. Please send us an email at openmv@openmv.io we'll get you a new one. Note the latest firmware is 3.4.1.

closed 2019-05-11 by iabdalkader

tpag26, 2019-05-12:
Have updated to latest firmware and same issue still occurs. Confirming that all pins are connected. Will email openmv at the address you provided. Thanks for quick reply