openmv/openmv #532: IDE file copy issue

nezra, 2019-05-28:
### Steps to reproduc
1. create a custom file you'll be importing. a complete script isn't necessary.
2. use "#%" in the file line, such as this: 

print("\tSending param #%d is %d bytes long" % (i, len(param)))

3. Import a custom file into the script 
4. start the script, and you are presented with the option for openmv to upload the file to the device. 

### Expected behaviour
library is uploaded to the device, unaltered.

### Actual behaviour
because the IDE strips all of the comments and extra spaces out, it shortens the line above to:

print("\tSending param 

which breaks the rest of the code. Since the IDE attempts to version check it both ways, any time an included file is different from one or the other device, you either have to do it manually by copying/pasting both ways before starting the script.

### Describe your setup

**IDE version:**
2.2.0

**Firmware version:**
3.4.1

**Operating system:**
Windows 10

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2019-05-28:
This is a legit issue. I will have to make the comment parser better.

kwagyeman, 2019-05-28:
I'm not sure how to solve this trivially however. I'll have to make code that walks each line and keeps a stack context of what is going on. I do this in other parts of the IDE for the auto complete code, but, not for the comment stripping.

iabdalkader, 2019-05-28:
@kwagyeman Can you use this parser in the IDE ? https://github.com/python/cpython/blob/master/Parser/parser.c

nezra, 2019-05-28:
Would it be possible to provide just an option to copy the file over as it is? i'm not condemning anything here, as i appreciate all of the work

I've found another curious behavior that I can toss into another issue if needed, but it in a way relates to what @iabdalkader mentions. I don't understand why the libraries need to be run through a parser at all forced or as an option. The reason is while the IDE looks for files in libraries folders, if you have a file structure like:

library_name_folder
    Library_file_1
    Library_file_2

When the script calls for the file via

import library_name_folder.Library_file_1

the IDE copies/parses the file and puts it in the root directory on the SD card, instead of recreating the folder structure. This makes the script throw an error because it's looking in the wrong place for it. the script being ran is looking for the folder structure, whereas the IDE parses it to the root directory of the SD card.

The point is, why is the code being parsed at all, instead of just copied to the SD card? in an internal storage situation i could see there to be logic for it, but, as i can simply copy a file with the extra spaces & comments onto the SD card i don't see the point. I know the code runs, as while the IDE parses through the library file, it doesn't step through everything.

Using the above folder/file list, If the main script calls for library_file_1; and library_file_1 includes Library_file_2; the IDE only parses library_file_1 onto the SD card. I still need to manually copy over file 2. 

so, back to my initial point, why are the libraries being parsed at all instead of just copied over 1:1?

on the "trivially point" you made above, if there is a reason i don't understand, wouldn't the only scenario where a # be is in the middle of a string? otherwise, everywhere else in in the code a # is treated as a comment.

nezra, 2019-05-28:
I should also mention, if you use the ide's "Tool" menu item "save open script to openmvcam" option it gives you a choice to strip the comments and extra spaces out. if you choose "no" it copies the file 1:1 over without breaking it. the only time it breaks is if you when the parser runs it, gives you the error, and automatically does the stripping.

kwagyeman, 2019-05-28:
Yeah, the comment stripping right now in the IDE is just a regex for a # in a ine. It doesn't handle # inside of "".

As for the library structure. I'm not sure if I added support for folders of modules to the IDE to copy over. I don't think it's that smart right now. It kinda assumes you have a module per file. This is because the IDE has to deal with folder structures that are nested looking for modules. It can't really look for modules that are composed of a folder structure right now.

nezra, 2019-05-28:
A a quick and dirty solution then, can you just add the choice to strip files like with the "tool" menu item for scripts running instead of it deciding to do the regex with no option?

As for the library structure, the IDE does check subfolders, as that's where i was putting libraries i was referencing. I started doing this to clean up the workspace after i read the blog post where you added it.

https://openmv.io/blogs/news/openmv-ide-v1-8-1-is-here-w-firmware-v2-7-0
http://wiki.micropython.org/Importing-Modules

It doesn't matter if its in either format:

Library
  Library_file

or

Libraries
  Library
    Library File

It just doesn't recreate them when the files are copied over. This causes the script to fail because a file doesn't exist in the same folder type path.

I thought it was worth mentioning since we don't really need to parse libraries based on how i worked around it. The library issue is completely separate. That's more of an annoyance since you wind up with up to 3 copies, or rewriting the libraries themselves to accomodate the single level setup. i can shoot you the code in question if you'd rather see the issue in a zip file with the folder structure and the resulting openmvcam flash drive if needed.

kwagyeman, 2019-05-29:
Yeah, so, for various reasons this feature can't possibly work unless you keep the hierarchy flat. I can't support allowing you to put libraries in any folder and structured library layout at the same time.

Please write you code to with one module per file and don't put paths in the import lines.

...

I don't really have a fix for allowing this to support you having paths in imports and importing modules that are folders.