openmv/openmv #534: Binary ops bug when using BMP

iabdalkader, 2019-06-06:
### Steps to reproduce
1. Create a 320x240 black filled BMP and as mask.bmp
2. Run the following script:
`import sensor, image
sensor.reset()
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time=1000)
while(True):
   sensor.snapshot().b_and("mask.bmp")`

### Expected behaviour
The framebuffer should be all black/clear

### Actual behaviour
Garbage output black/white dots.

### Describe your setup

**IDE version:**
2.2.0
**Firmware version:**
3.4.1/2
**Operating system:**
Linux
### For developers

**GCC version:**
7.3.1 7-2018-q2-update
**Build target:**
OPENMV3/4

iabdalkader, 2019-06-06:
@kwagyeman

kwagyeman, 2019-06-06:
Will fix.

commit 5b529521 (2019-09-08): Fix binary ops bug.

* fix #534
PR #583 (2019-09-09): Fix binary ops bug.
* fix #534
closed 2019-09-09 by iabdalkader