openmv/openmv #546: Wakeup issue in OpenMV H7

xumengwei, 2019-07-01:
It seems like there's a wakeup issue in OpenMV H7 camera.
`def capture_img():
    pass
def test_led():
    intvl = 0.5
    red_led.on()
    utime.sleep(intvl)
    red_led.off()
    green_led.on()
    utime.sleep(intvl)
    green_led.off()
    blue_led.on()
    utime.sleep(intvl)
    blue_led.off()

rtc = pyb.RTC()
rtc.wakeup(2000, capture_img)

while (True):
    test_led()
    pyb.stop() # sleeping mode, 500 uA`

This code doesn't wake up the MCU. According to [this](http://forums.openmv.io/viewtopic.php?f=3&t=1387&p=8590#p8590), it works on F7 but not H7. Please help fix it.

iabdalkader, 2019-08-13:
Fixed in #561

iabdalkader, 2019-08-13:
@xumengwei Please open another issue for the datetime bug.

closed 2019-08-13 by iabdalkader