openmv/openmv #558: PX4 Cant recognize Mavlink Optical Flow example.

tech0x, 2019-07-27:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

tech0x, 2019-07-28:
I made some debug work and find these:
1. PX4 handle only OPTICAL_FLOW_RAD via mavlink MSGID 106 instead OPTICAL_FLOW MSGID 00.
2. But structure OPTICAL_FLOW_RAD to different.
OPTICAL_FLOW_RAD:
uint32_t integration_time_us; /*< [us] Integration time. Divide integrated_x and integrated_y by the integration time to obtain average flow. The integration time also indicates the.*/
float integrated_x; /*< [rad] Flow around X axis (Sensor RH rotation about the X axis induces a positive flow. Sensor linear motion along the positive Y axis induces a negative flow.)*/
float integrated_y; /*< [rad] Flow around Y axis (Sensor RH rotation about the Y axis induces a positive flow. Sensor linear motion along the positive X axis induces a positive flow.)*/
float integrated_xgyro; /*< [rad] RH rotation around X axis*/
float integrated_ygyro; /*< [rad] RH rotation around Y axis*/
float integrated_zgyro; /*< [rad] RH rotation around Z axis*/
uint32_t time_delta_distance_us; /*< [us] Time since the distance was sampled.*/
int16_t temperature; /*< [cdegC] Temperature*/

OPTICAL_FLOW:
float flow_comp_m_x; /*< [m] Flow in x-sensor direction, angular-speed compensated*/
float flow_comp_m_y; /*< [m] Flow in y-sensor direction, angular-speed compensated*/
int16_t flow_x; /*< [dpix] Flow in x-sensor direction*/
int16_t flow_y; /*< [dpix] Flow in y-sensor direction*/
float flow_rate_x; /*< [rad/s] Flow rate about X axis*/
float flow_rate_y; /*< [rad/s] Flow rate about Y axis*/

How I can correctly translate values to OPTICAL_FLOW_RAD format ?

coolsyn, 2019-09-01:
when I ran AprilTag landing script in openmv H7, QGroundControl can not receive msg id = 149, but it can get message id =132, which confused me.

iabdalkader, 2019-09-06:
@kwagyeman

kwagyeman, 2019-09-06:
I don't really know what the problem is. Please ask Randy Mackaday.

iabdalkader, 2019-09-08:
@rmackay9 Can you help with this issue ?

rmackay9, 2019-09-08:
@iabdalkader, I'm afraid not.  PX4 is completely different software.  ArduPilot is the firmware that I used.

[ArduPilot's OpenMV for optical flow wiki page is here](http://ardupilot.org/copter/docs/common-openmv-optflow.html) which includes setup info for the camera.  [AP's precision landing page is here ](http://ardupilot.org/copter/docs/precision-landing-with-irlock.html)although it doesn't include OpenMV information the more general setup info is there.

I think that it's probably best to close this issue because it's not really an openmv issue.

closed 2019-09-08 by kwagyeman