openmv/openmv #563: Rotate image from sensor 90 degrees.

FPSychotic, 2019-08-22:
Hi, I have a small 320x240 display behind the camera, (as a digital camera) oriented in in horizontal and I cannot rotate it to fit it to the camera orientation that is fitted in vertical, (that i cannot rotate either), so to fit both for a natural view I would like rotate the image 90 degrees, is not important if I need crop the image to a square format as 240x240. there is some command to do it? Im begginer in everything,cameras,python, so sorry iid it is a weird question.

thank you

kwagyeman, 2019-08-22:
Hi, can you ask this question on the forum? Github is for firmware bug issues.

Anyway, do:

sensor.set_windowing((240,240))

And then when you have an image use the set() method to rotate it by 90. See the API.

FPSychotic, 2019-08-22:
oohh, really Im sorry the inconvenience. Im quite new in all this things git, cameras,cv ,coding and I get confused often. thank you  a lot that Im sure will help.

closed 2019-08-22 by FPSychotic