openmv/openmv #564: AprilTag with vertical flip does not work.

thyrrestrup, 2019-08-23:
### Steps to reproduce
Open `find_aprils.py` and add the following `sensor.set_vflip(True)`

### Expected behaviour
Detects AprilTags as usual.

### Actual behaviour
Does not detect AprilTags.

### Describe your setup

**IDE version:**
2.2
**Firmware version:**
3.4.1
**Operating system:**
Mac OS X
### For developers

**GCC version:**

**Build target:**

kwagyeman, 2019-08-23:
Unless the tag is also flipped this wouldn't work. The tag is a rigid structure with a certain orientation. Flipping the image would flip the code of the tag removing it from the right tag family.

That said, I can modify the tag decoder logic to handle this by also checking vertically flipped and horizontally mirrored patterns. Would you like that? This isn't too hard.

thyrrestrup, 2019-08-26:
That sounds great - is there a performance hit to this? I think it's counter-intuitive to have to flip the tag when the camera is mounted upside down.

thyrrestrup, 2019-08-27:
Is it possible to set a flag for h_mirror and v_flip and then still only run through the lookup 4 times?

kwagyeman, 2019-09-03:
Fixed https://github.com/openmv/openmv/pull/576

closed 2019-09-03 by kwagyeman

thyrrestrup, 2019-09-03:
Sweet!