openmv/openmv #567: Simulate OpenMV input stream for development

cansik, 2019-08-24:
This is more a feature request than an actual issue with the system. I really enjoy working with OpenMV and it helps me to create installations which would not be possible without such a small camera & board.

It's great that I can just plugin the camera and see the live framebuffer output. While this is convenient when developing onsite, it is sometimes not possible to develop and test in a productive environment. So it would be great if there is a possibility to run OpenMV with a pre-recorded video as a camera input.

There are different ways to achieve this, either by emulating the whole platform (quemu) or to stream a video stream to the camera, use this frame as an input buffer and let the connected OpenMV run its python code.

I think the second option would be simpler to implement, regarding bandwidth, the video could even be stored on the SD card of the camera. The idea is just to have a way to work with pre-recorded footage for testing and development.

Have you ever thought about something like this?

kwagyeman, 2019-08-24:
Hi, this feature has been available for a while. See the ImageWriter and ImageReader class. They allow you to record the frame buffer losslessly to the SD card and then play it back later at whatever time you'd like. You still need to use the OpenMV Camera to do this but you can use another one anywhere.

kwagyeman, 2019-08-24:
OpenMV IDE can convert these files also to mp4s if you want to put them in another program.

closed 2019-09-02 by kwagyeman