openmv/openmv #617: compilation error during modbtree.c build

crazyuav, 2019-10-23:
Hi,
I referenced the webpage <https://github.com/openmv/openmv/wiki>, and make the firmware.

```
openmv version:
openmv at v3.4.3
micropython@7747c5b

arm-none-eabi-gcc version:
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q3-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

```

I got the some compilation errors as below:

```
..........
CC ../../extmod/modbtree.c
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:1:1: error: expected identifier or '(' before '.' token
 ../../include/db.h
 ^
In file included from ../../extmod/modbtree.c:38:0:
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: data definition has no type or storage class [-Werror]
 } PAGE;
   ^~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: type defaults to 'int' in declaration of 'PAGE' [-Werror=implicit-int]
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:127:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* page number stored on */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:157:2: error: unknown type name 'recno_t'
  recno_t nrecs;   /* number of records */
  ^~~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:158:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* page number stored below */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:247:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* the page number */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:248:2: error: unknown type name 'indx_t'
  indx_t index;   /* the index on the page */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:252:2: error: expected specifier-qualifier-list before 'PAGE'
  PAGE *page;   /* the (pinned) page */
  ^~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:282:2: error: unknown type name 'DBT'
  DBT  key;   /* B: Saved key, or key.data == NULL. */
  ^~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:283:2: error: unknown type name 'recno_t'
  recno_t  rcursor;  /* R: recno cursor (1-based) */
  ^~~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:310:2: error: unknown type name 'MPOOL'
  MPOOL  *bt_mp;  /* memory pool cookie */
  ^~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:312:2: error: unknown type name 'DB'
  DB  *bt_dbp;  /* pointer to enclosing DB */
  ^~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:315:2: error: expected specifier-qualifier-list before 'PAGE'
  PAGE  *bt_pinned;  /* page pinned across calls */
  ^~~~
In file included from /usr/arm-none-eabi/include/stdio.h:35:0,
                 from ../../extmod/modbtree.c:27:
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:36:22: error: unknown type name 'DB'
 int  __bt_close __P((DB *));
                      ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:37:35: error: unknown type name 'DBT'
 int  __bt_cmp __P((BTREE *, const DBT *, EPG *));
                                   ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:39:29: error: unknown type name 'DBT'
 int  __bt_defcmp __P((const DBT *, const DBT *));
                             ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:39:42: error: unknown type name 'DBT'
 int  __bt_defcmp __P((const DBT *, const DBT *));
                                          ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:40:32: error: unknown type name 'DBT'
 size_t  __bt_defpfx __P((const DBT *, const DBT *));
                                ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:40:45: error: unknown type name 'DBT'
 size_t  __bt_defpfx __P((const DBT *, const DBT *));
                                             ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:41:29: error: unknown type name 'DB'
 int  __bt_delete __P((const DB *, const DBT *, u_int));
                             ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:41:41: error: unknown type name 'DBT'
 int  __bt_delete __P((const DB *, const DBT *, u_int));
                                         ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:42:37: error: unknown type name 'DBT'
 int  __bt_dleaf __P((BTREE *, const DBT *, PAGE *, u_int));
                                     ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:42:44: error: expected declaration specifiers or '...' before 'PAGE'
 int  __bt_dleaf __P((BTREE *, const DBT *, PAGE *, u_int));
                                            ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:43:25: error: unknown type name 'DB'
 int  __bt_fd __P((const DB *));
                         ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:44:30: error: expected declaration specifiers or '...' before 'PAGE'
 int  __bt_free __P((BTREE *, PAGE *));
                              ^
```

Thanks!

BR  Neal

iabdalkader, 2019-10-23:
I can't say for sure, maybe the build was broken at that point. But micropython should be at branch openmv or openmv-1.9.4.
If you clone the repo without changing anything it should build fine.

```
git clone --recursive https://github.com/openmv/openmv.git
cd openmv/src
make TARGET=OPENMV3
```

crazyuav, 2019-10-24:
I clone Repo in windows,and there're some symlink,so generate errors;
After clone repo in ubuntu,"modbtree.c build" is work,but still anothers errors as below:
```
make[1]: Entering directory '/home/bbb/tools/openmv/src/bootloader'
CC src/stm32fxxx_hal_msp.c
CC src/usbd_conf.c
CC src/usbd_core.c
CC src/usbd_ioreq.c
CC src/usbd_cdc_interface.c
CC src/usbd_ctlreq.c
CC src/stm32fxxx_it.c
CC src/string0.c
CC src/flash.c
CC src/usbd_cdc.c
CC src/main.c
CC src/usbd_desc.c
make[1]: Leaving directory '/home/bbb/tools/openmv/src/bootloader'
/usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: cannot open output file /home/bbb/tools/openmv/src/build/bin/bootloader.elf: No such file or directory
collect2: error: ld returned 1 exit status
Makefile:556: recipe for target 'bootloader' failed
make: *** [bootloader] Error 1

```

iabdalkader, 2019-10-24:
Oh yes this is bug in the Makefile makes non-parallel builds fail, I'll fix it soon. For now just add -jn (n== number of CPUs).

ex: `make -j4 TARGEt=OPENMV3`

iabdalkader, 2019-10-25:
I'll close this when it gets fixed.

iabdalkader, 2019-10-26:
Fixed in #625

closed 2019-10-26 by iabdalkader