openmv/openmv #641: Canny edge detector always finds 2 vertical lines independent of image

gvcp, 2019-11-17:
When trying to find edges with CANNY in QVGA format, it always finds 2 vertical lines 
at the right of the image. The number of white pixels differ between 475 and 476 white pixels from frame to frame. Looks like a rectangle consisting of 2 parallel lines each 1 pixel wide with a space of 1 black pixel between them. The upper part of the rectangle is always closed, the lower part changes between frames. (sometimes open sometimes closed so I get 475 or 476 pixels)
See attached screenshot.

The problem does not appear when using CANNY whith QQVGA format and also doesn´t with EDGE_SIMPLE when using QVGA or QQVGA

[image]

### Steps to reproduce
Execute the code below

### Expected behaviour
Should do what it does but without the vertical lines.

### Actual behaviour
CANNY always finds these lines, no matter which image is used.
It also does with a black image (lens covered)

### Setup

**IDE version: 2.2.0

**Firmware version: 3.5.0

**Operating system: MacOS 10.13.6

##Code:
import sensor, image, time

sensor.reset() # Initialize the camera sensor.
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000) 
sensor.set_gainceiling(8)

clock = time.clock() # Tracks FPS.
while(True):
    clock.tick() # Track elapsed milliseconds between snapshots().
    img = sensor.snapshot() 
    img.find_edges(image.EDGE_CANNY, threshold=(40, 90))

kwagyeman, 2019-11-17:
Thanks for finding this bug.

iabdalkader, 2019-12-01:
Fixed

closed 2019-12-01 by iabdalkader