openmv/openmv #642: Neural Nets information horribly out of date

kutenai, 2019-11-20:
### Steps to reproduce
1. Just try and work through the "example" for nn
2.
3.

### Expected behaviour
It might follow the examples a little bit

### Actual behaviour
I spent an entire day trying to this this to all compile, but there are so many issues.
It depends on 'caffe' which has had practically no commits since 2016. So, it's very difficult/impossible to build that system without applying code patches, digging into the source code, etc. It's very difficult.

I just bought the newest H7 version of your device, but these new "nn" features are just useless without a proper workflow to build something. I'm very dissapointed that your documentation is relying on a 3+ year old out of date library for the more advanced features.

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

iabdalkader, 2019-11-20:
Our code is based on ARM's CMSIS_NN code and examples which use Caffe, so this wasn't really our first choice, however Caffe builds fine on my machine (Arch Linux so it's always up to date) and other users' machines, if you're having issues building it you may want to try on an Ubuntu machine/VM. That said, it is an outdated framework and I understand and share your frustration...The good news is we have two other options for AI, [support for models generated by CUBE.AI](https://github.com/openmv/openmv/blob/master/src/stm32cubeai/README.md) have been recently added (Note this require a firmware build, and supports a single/static model) and we have TensorFlow Lite support (Not sure if it's fully documented yet, @kwagyeman should know more about this).

kwagyeman, 2019-11-20:
Hi, we don't advertise the CNN features on the product page because folks fail to be able to install the caffe library. It took me a solid weekend to install caffe correctly. Our documentation does indeed work if you are actually able to accomplish this and run caffe. That said, it's basically impossible to do this unless you have a clean Linux box to start with, a GPU, and a weekend to burn.

Anyway, the CNN features aren't something that's been what you'd buy the system for. We just released TensorFlow micro support finally now which changes this. The previous CMSIS-NN support was released by ARM last year and abandoned basically after release. We tried to get it working and showed that it could run on our system but we didn't receive any help from ARM or more support on this effort after getting a demo out.

kutenai, 2019-11-20:
I am using a mac, so I tried to install it there.
I also tried to install on an Ubuntu 18.04 VM, but this also failed. The 
instructions are old, and no longer apply. Also, caffe no longer 
compiles against OpenCV 4, which is the default on all of the platforms. 
I tried to install opencv@2, and that also caused an error, perhaps I'd 
have to *clean* all other files or something. To be honest, it wasn't 
and isn't worth the headaches. Requiring caffe is a big issue, and I 
don't know how much luck you would have building on that platform 
*today*. It might have worked a couple of years ago for you, but if you 
tried it again, I bet you'd have issues also.

If you ARE able to make it work, I'd love to see updated instructions.

yes, I saw the tensorflow lite support, but the documentation is *very* 
limited. I never saw the other library at all, so did not try and use it.
I wanted to try some things out without digging too deeply into the 
code, so I'm not really in a position to dig into the firmware for 
CMSIS_NN support at this time.

Kwabena W. Agyeman wrote on 11/19/19 5:51 PM:

kwagyeman, 2019-11-20:
Yeah, I'd not bother with Caffe anything anymore. I'll delete all that stuff once we port the demo models we made. As for TensorFlow, see this doc: https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/experimental/micro/examples/person_detection/training_a_model.md

The output .tflite file before you turn it into a c file is runnable on the camera.

Note that as mentioned elsewhere on the forums the error messages from TensorFlow appear in the console before the assert is thrown and displayed by the IDE. So, you need to look at the serial console for the error message. The one the IDE displays is just our code noting that TensorFlow had an error.

kwagyeman, 2020-07-18:
Hi, all this stuff was removed. We are TensorFlow Lite for Microcontrollers 100% now.

closed 2020-07-18 by kwagyeman