openmv/openmv #651: Version 3.4 works normally But 3.5 Exception

Kevincoooool, 2019-12-10:
this is my own made board STM32H743VIT6 and MT9V034

### Steps to reproduce
1.My own OPENMV4 board use STM32H743VIT6 and MT9V034
2.When I use firmware before 3.4.1, all functions are normal, and the sensor cannot be initialized after switching to 3.5 firmware
3.Flash to 3.5 firmware ，the ERROR.LOG prompt 
FATAL ERROR:
Failed to init sensor, error: -3
and White light flashes quickly

### Expected behaviour
Tell us what should happen

### Actual behaviour
3.5 firmware abnormal

### Describe your setup

**IDE version:2.2

**Firmware version:3.5

**Operating system:win10

### For developers

**GCC version:**

**Build target:**

iabdalkader, 2019-12-10:
The "official" H7 cam + MT sensor is working fine, I don't know what's wrong with your board you'll have to figure that out on your own, if it helps -3 means the sensor I2C address is not recognized.

closed 2019-12-10 by iabdalkader

Kevincoooool, 2019-12-11:
> The "official" H7 cam + MT sensor is working fine, I don't know what's wrong with your board you'll have to figure that out on your own, if it helps -3 means the sensor I2C address is not recognized.

what's different with 3.4 and 3.5

kwagyeman, 2019-12-11:
Literally everything. We updated MicroPython and fixed a bug of bugs.