openmv/openmv #652: Memory Error in  USB VCP mode

tech0x, 2019-12-10:
1. I try to use OpenMV H7 as camera but device freezes with example usb code after several seconds after runing.

import sensor, image, time, ustruct
from pyb import USB_VCP

usb = USB_VCP()
sensor.reset()                      # Reset and initialize the sensor.
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)     # Wait for settings take effect.

while(True):
    cmd = usb.recv(4, timeout=5000)
    if (cmd == b'snap'):
        img = sensor.snapshot().compress()
        usb.send(ustruct.pack("<L", img.size()))
        usb.send(img)

iabdalkader, 2019-12-10:
Are you running this from the IDE ? If so you shouldn't, the IDE uses the USB devices

tech0x, 2020-01-07:
I tests without ide, how described

iabdalkader, 2020-01-18:
I just ran this example and it seems to be working fine, are you running the host scripts in the comments ? Also please fill in the issue template (camera, firmware version, os etc...).

iabdalkader, 2020-01-30:
Please reopen if there's an update.

closed 2020-01-30 by iabdalkader