openmv/openmv #660: WiFi.isconnected() not working

chrisbuerginrogers, 2020-01-01:
### Steps to reproduce
WiFi = network.WINC() causes WiFi.isconnected() to be True - even before WiFi.connect has been executed (whereas on the pyboard, WiFi.isconnected is false until WiFi.connect has been successfully run)

>>> WiFi = network.WINC()
Running in Station mode...
>>> WiFi.isconnected()
True
>>> 

### Expected behaviour
WiFi.isconnected() should be false until it is connected

### Actual behaviour
it is true right away

### Describe your setup

**IDE version:**
NA

**Firmware version:**
MicroPython v1.11-omv OpenMV v3.5.2 

**Operating system:**
mac

### For developers

**GCC version:**

**Build target:**

iabdalkader, 2020-01-18:
I can reproduce this issue
```
MicroPython v1.11-omv OpenMV v3.5.2 2019-12-17; OPENMV4-STM32H743
Type "help()" for more information.
>>> import network
>>> wifi = network.WINC()
Running in Station mode...
>>> wifi.i
ifconfig        isconnected
>>> wifi.isconnected()
False
>>> 
```

iabdalkader, 2020-01-24:
Please reopen the issue if there's an update.

closed 2020-01-24 by iabdalkader

chrisbuerginrogers, 2020-02-16:
I cannot seem to replicate it any more either.  Sorry to waste your time.