openmv/openmv #674: OSError: Failed to init the MLX90640!

RawanAlharbi, 2020-01-22:
This problem only occurs with MLX. I am not facing any problems with the grideye. 

### Steps to reproduce
1. Connect sparkfun MLX90640 to  P4 (SCL), P5 (SDA), GND and 3.3V on OpenMV H7 OV7725
2. open any of the MLX90640 demos and run. 

### Expected behaviour
I should see the thermal image obtained from MLX90640.

### Actual behaviour
"OSError: Failed to init the MLX90640!"

### Describe your setup

**IDE version:**  2.2.0

**Firmware version:** I don't know

**Operating system:** tested on 2 laptops one is macOS catalina and the other is windows 10

### Links:
posting.php?f=6&t=1704
viewtopic.php?f=6&t=1143

iabdalkader, 2020-01-23:
Hi, I just tested my MLX90640 and it seems to be working fine, please double check the wiring. With the camera lens pointing towards you, GND should be connected to the first pin on the right header, the 3.3v connected to the first pin on the left header, SDA and SCL should be connected to the 3rd and 4th pins on the left header respectively, if it still doesn't work you should replace the wires. Note you can also scan the I2C bus to make sure the sensor is connected and the camera can talk to it, the following script should print ```[51]``` which is the sensor's I2C address.
```
from pyb import I2C

i2c = I2C(2, I2C.MASTER) # The i2c bus must always be 2.
print(i2c.scan())
```

[image]

RawanAlharbi, 2020-01-24:
It appears that the problem was with the MLX itself. I tried it with another MLX today and it worked! Thanks!

closed 2020-01-24 by RawanAlharbi