openmv/openmv #678: Problem with UART

halsted, 2020-01-29:
### Steps to reproduce
1. Using DSD Tech HC-06 Bluetooth module. Connect P0 to TX pin, P1 to RX pin. Power HC-06 with +5V, gnd from power supply. Set Baud rate in program below to 9600 baud.
[BT_config_orig.txt](https://github.com/openmv/openmv/files/4131188/BT_config_orig.txt)

2. Program works using old version of firmware on Cam M7 (and H7), but does not work using latest firmware version.  It works on Firmware version 3.2.0 and 3.4.3, but not the latest.

3.
### Expected behaviour
On a working OpenMV Cam M7 with firmware version 3.2.0, the output is as shown below, once the HC-06 baud rate has been configured to 115200. The HC-06 can be successfully configured and characters can be sent and received using a Bluetooth serial terminal.

MicroPython v1.9.4-4531-g65b367440 on 2018-11-04; OPENMV3 with STM32F765
Type "help()" for more information.
>>> AT
b'OK'
AT+VERSION
b'OKlinvorV1.8'
AT+PIN5555
b'OKsetPIN'
AT+NAME=LH_HC06
b'OKsetname'
AT+BAUD8
b'OK115200'
d
t

### Actual behaviour
Tell us what happens instead
MicroPython v1.11-omv OpenMV v3.5.2 2019-12-17; OPENMV3-STM32F765
Type "help()" for more information.
>>> AT
None
AT+VERSION
None
AT+PIN5555
None
AT+NAME=LH_HC06
None
AT+BAUD8
None

Using the latest firmware version, I cannot communicate with the DSD Tech HC-06 Bluetooth 2.0 module. It works with old firmware versions. This has been verified on multiple cameras.

### Describe your setup
P0 (UART 1 RX) -> HC-06 TX pin
P1 (UART 1 TX) -> HC-06 RX pin
Connect VCC of HC-06 to +5V, GND to gnd

**IDE version:**
2.2.0

**Firmware version:**
3.5.2 does not work. Old versions (3.4.3 and 3.2.0) work.

**Operating system:**
Windows 10 Pro

### For developers

**GCC version:**

**Build target:**

halsted, 2020-01-30:
So I may have a solution, but I'm not sure.
Could this issue be due to the Timer 4 problem discussed https://forums.openmv.io/viewtopic.php?f=6&t=1696

I was able to get the code above to work by using the firmware.bin in the link above. However, it doesn't seem to pass the self-test and blink the Blue LED. After waiting a long time, I'll unplug the camera and plug it back in and it doesn't recognize it. It thinks it's bricked and reloads the latest firmware. However, somehow it eventually worked.
I had to increase the timeout to 1000 ms in the uart init:
uart.init(115200, bits=8, parity=None, stop=1, timeout=1000, flow=0)

Does it make sense that uart 1 is affected by timer 4 being disabled?

Why does the self-test fail after updating a Cam M7 with the firmware.bin patch?
Is the issue the same with the Cam H7?
(I would post to a forum, but my account has not been activated yet.)
Thanks!

kwagyeman, 2020-01-30:
Probably related to MicroPython issues. You're in luck however. We can get this fixed for the next release which is coming soon.

iabdalkader, 2020-01-30:
I test the UART RX/TX with every release, it was working last time I checked, and it's not related to TIM4. Can you try adding this arg ```timeout_char = 100``` ?
I'm going to retest everything before the next release anyway.

iabdalkader, 2020-01-30:
I just tested 3.5.2 I can read/write from UART normally, please double check the wiring (or replace the wires) and try the timeout_char arg, also you should test with a USB to serial (if you have one).

halsted, 2020-01-30:
I just updated a Cam M7 from an old firmware version to 3.5.2. It works when I use timeout=1000. I didn't try timeout_char=100 yet. The timing is different between 3.5.2 and previous firmware versions because I did not need to use timeout with the old versions. So the uart does work, just with somewhat different timing.

kwagyeman, 2020-01-30:
I think we should default that timeout to 1000. I don't know why MP has it set so low since everyone runs into it being an issue.

halsted, 2020-01-30:
By the way, what is the sequence of steps to load the firmware.bin that fixes the Timer 4 problem? I haven't been able to get it to work.

kwagyeman, 2020-01-30:
Tools-> Run Bootloader, select the file and then program the camera.

halsted, 2020-01-30:
When I do that, it doesn't pass the self test. It always thinks the camera is bricked and I have to restore 3.5.2. Do I need to jumper Boot0? That doesn't seem to work either.

kwagyeman, 2020-01-30:
Yeah, that would imply your hardware is deficient... Regarding the self test, you can just delete the test.py script on the upy drive and the camera will skip it. As long as the sensor works the camera will be fine.

I.E flash the firmware, the upy drive should appear, then delete test.py.

If you are getting the flashing blinking white led then that's a more serious issue. That would imply your hardware is bad after a firmware update. In later versions we push clocks faster, etc. More performances, 400 MHz -> 480 MHz. Etc.

halsted, 2020-01-30:
OK - Got it. It fails on multiple OpenMV3 R1 (M7) from 10/15/2016 but it works for OpenMV-H7 R1 from 10/27/18. I guess for the old M7 devices, we should just use an old firmware version such as 3.4.3 which doesn't have the timer 4 problem. For the new H7, we can use the patch to 3.5.2. Thanks,

kwagyeman, 2020-01-30:
M7 Firmware shouldn't be broken. I'll flash one and see what's up when I have time. Sorry, it's hard keeping all this stuff update to date. We need more help from the community.

halsted, 2020-01-30:
I haven't verified the timer 4 issue on M7 firmware. Some students reported the timer problem, but they are probably using H7.

iabdalkader, 2020-01-30:
We're about to release the new firmware anyway, so the TIM4 issue will be resolved soon. I'm going to close this issue now.

closed 2020-01-30 by iabdalkader