openmv/openmv #691: OpenMV Wave file output

eddiekim95, 2020-02-06:
TLDR: code works on older model (OpenMV3 R1) but not on newer model (OpenMV H7)

### Steps to reproduce
1. A speaker connected to the Adafruit Class D amplifier (https://www.adafruit.com/product/2130). A+ pin connects to the DAC pin on the OpenMV cam (P6). A- pin connects to ground. SD pin is left open. Vin pin is connected either to 3.3V supply from OpenMV cam or 5V from external power supply. Gnd pin is connected to ground on OpenMV cam. 
2. Have the associated libraries [(wave.py,](http://micropython.org/resources/examples/wave.py) [chunk.py)](http://micropython.org/resources/examples/chunk.py) , and put them at the top level of the OpenMV cam directory. Link to [test.wav](http://micropython.org/resources/examples/test.wav) file.
3. Run the audio output test code that should read a wave file and output it to the DAC. 

### Expected behaviour
Tell us what should happen:
The OpenMV cam should read the test.wav file and output to the DAC to be amplified by the class D amplifier. 

### Actual behaviour
Tell us what happens instead:
On an older model of the OpenMV cam (OpenMV3 R1), the code works fine and the audio plays.
On the new OpenMV cam H7, the audio does not play. Here are some of the things I have tested to work:
- tested the function of the DAC on the board by outputting a sin wave [works]
- tested the function of the wave library required to read .wav files [works]
- tested the output of a signal generated and output through the DAC and amplifier, will play a sin wave [works]
- tested the amplifier with a 5V supply voltage and 3.3V supply voltage [both work]

### Describe your setup
The amplifier and speaker are connected to the OpenMV H7 and the code is run on the OpenMV IDE. I followed this tutorial and got the wave.py, chunk.py, and sample wave file from this site. https://docs.micropython.org/en/latest/pyboard/tutorial/amp_skin.html

Test Code:
`import time, pyb, wave, gc
from pyb import DAC

dac = DAC(pyb.Pin("P6"))

def sound():
    gc.collect()
    f = wave.open('test.wav', 'rb')
    dac.write_timed(f.readframes(f.getnframes()), f.getframerate())
    print(f.getnframes())
    print(f.getframerate())
    #f.close()
    del f
    gc.collect()

while(True):
    sound()
    time.sleep(6000)
    pyb.hard_reset()
`

**IDE version:**
OpenMV IDE 2.2.0
**Firmware version:**
3.5.2
**Operating system:**
Windows 10
### For developers

**GCC version:**

**Build target:**

iabdalkader, 2020-02-06:
Hi, thanks for taking the time to report this bug. I've already found the issue with the DAC driver and I'm working on it. This fix will most likely be in the next release.

iabdalkader, 2020-02-07:
Fixed.

closed 2020-02-07 by iabdalkader