openmv/openmv #694: CIF settings for OV7725 ?

labviewegypt, 2020-02-09:
To get CIF images,is it settings we have to send it to the camera or its done over MCU itself ?

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2020-02-09:
Set the resolution to CIF.

labviewegypt, 2020-02-09:
@kwagyeman 
Thank you for you reply
I want to clear more about my question , i am using my own board with stm32l496 mcu ,i just using openMV  ov7725 camera breakout over my PCB .
Also,i am using the drivers from openMV repository,Everything working fine with me.
I want to take snapshots with CIF resolution and i dont know how ?
My question is there any certain configurations i have to send it to the camera to get this CIF resolution ? or to get CIF images you have to get images in VGA resolution (for example) then do some operation on MCU to get CIF ?

kwagyeman, 2020-02-09:
You do sensor.set_framesize(sensir.cif) in MicroPython. If you are trying to do this in C please lookup what the above MicroPython snippet does and mimic that in C. Please see the sensor_set_framesize() function.

Also, if you are using your own board we do not provide help support for that.

closed 2020-02-09 by kwagyeman