openmv/openmv #733: PORTENTA with stm32h743_af.csv

Dieter-Graef, 2020-02-27:
[stm32h747_af.zip](https://github.com/openmv/openmv/files/4261353/stm32h747_af.zip)

Hello,
I just noticed that you are working an the PORTENTA device but using the file stm32h743_af.csv for pin assignment. Therefor i want kindly point you to my Gihub project https://github.com/Dieter-Graef/Pinmaker that produces the af.csv files right from the STM32CubeMX database

. It requires a bit work after. Mainly resorting and renaming some pins. See the examples with a stm32h747_af.csv file that works fine.
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

### For developers

**GCC version:**

**Build target:**

kwagyeman, 2020-02-28:
What's the issue being fixed?

Dieter-Graef, 2020-02-28:
LCD with DSI cant be used.

iabdalkader, 2020-03-06:
Thank you for the AF file, there's no driver for DSI at this point anyway. Note the H747 support is not official yet, when it becomes supported and the AF is updated I'll be sure update ours.

closed 2020-03-06 by iabdalkader