openmv/openmv #743: btree Error

skylin008, 2020-03-18:
### Steps to reproduce
1. make
2. show as follow error:
Warning: FROZEN_MPY_DIR is deprecated in favour of FROZEN_MANIFEST
CC ../../extmod/modbtree.c
In file included from ../../extmod/modbtree.c:37:0:
../../lib/berkeley-db-1.xx/PORT/include/db.h:1:1: error: expected identifier or '(' before '.' token
 ../../include/db.h
 ^
In file included from ../../extmod/modbtree.c:38:0:
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: data definition has no type or storage class [-Werror]
 } PAGE;
   ^~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:89:3: error: type defaults to 'int' in declaration of 'PAGE' [-Werror=implicit-int]
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:127:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* page number stored on */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:157:2: error: unknown type name 'recno_t'
  recno_t nrecs;   /* number of records */
  ^~~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:158:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* page number stored below */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:247:2: error: unknown type name 'pgno_t'
  pgno_t pgno;   /* the page number */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:248:2: error: unknown type name 'indx_t'
  indx_t index;   /* the index on the page */
  ^~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:252:2: error: expected specifier-qualifier-list before 'PAGE'
  PAGE *page;   /* the (pinned) page */
  ^~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:282:2: error: unknown type name 'DBT'
  DBT  key;   /* B: Saved key, or key.data == NULL. */
  ^~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:283:2: error: unknown type name 'recno_t'
  recno_t  rcursor;  /* R: recno cursor (1-based) */
  ^~~~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:310:2: error: unknown type name 'MPOOL'
  MPOOL  *bt_mp;  /* memory pool cookie */
  ^~~~~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:312:2: error: unknown type name 'DB'
  DB  *bt_dbp;  /* pointer to enclosing DB */
  ^~
../../lib/berkeley-db-1.xx/PORT/include/../../btree/btree.h:315:2: error: expected specifier-qualifier-list before 'PAGE'
  PAGE  *bt_pinned;  /* page pinned across calls */
  ^~~~
In file included from /usr/arm-none-eabi/include/stdio.h:35:0,
                 from ../../extmod/modbtree.c:27:
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:36:22: error: unknown type name 'DB'
 int  __bt_close __P((DB *));
                      ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:37:35: error: unknown type name 'DBT'
 int  __bt_cmp __P((BTREE *, const DBT *, EPG *));
                                   ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:39:29: error: unknown type name 'DBT'
 int  __bt_defcmp __P((const DBT *, const DBT *));
                             ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:39:42: error: unknown type name 'DBT'
 int  __bt_defcmp __P((const DBT *, const DBT *));
                                          ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:40:32: error: unknown type name 'DBT'
 size_t  __bt_defpfx __P((const DBT *, const DBT *));
                                ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:40:45: error: unknown type name 'DBT'
 size_t  __bt_defpfx __P((const DBT *, const DBT *));
                                             ^
../../lib/berkeley-db-1.xx/PORT/include/../../btree/extern.h:41:29: error: unknown type name 'DB'
 int  __bt_delete __P((const DB *, const DBT *, u_int));
                             ^
[... 425 lines cut ...]
3.

### Expected behaviour
Tell us what should happen

How to resolve this issue.

### Actual behaviour
Tell us what happens instead

### Describe your setup

**IDE version:**

**Firmware version:**

**Operating system:**

Linux phonelink-Ubuntu 4.15.0-74-generic #83~16.04.1-Ubuntu SMP Wed Dec 18 04:56:23 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

### For developers

**GCC version:**
gcc version 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] (GNU Tools for Arm Embedded Processors 7-2018-q3-update)
**Build target:**

iabdalkader, 2020-03-21:
The firmware builds fines with all the supported targets, there's an automated build that runs before each commit and it reports errors like these. I'm not sure what's wrong but it's a problem on your end for sure.

Meekdai, 2020-03-31:
The same problem, Is anyone solving？

kwagyeman, 2020-03-31:
You might need to git submodule update --init

closed 2020-03-31 by kwagyeman

frank26080115, 2020-08-11:
I encountered this problem and have a solution

Open the directory `openmv\src\micropython\lib\berkeley-db-1.xx\PORT\include`

See some of those `.h` files? For example `db.h`

The contents of that file is

    ../../include/db.h

Which is confusing the compiler

    ../../lib/berkeley-db-1.xx/PORT/include/db.h:1:1: error: expected identifier or '(' before '.' token
    ../../include/db.h

The solution is to change the contents of that file to

    #include "../../include/db.h"

There will be multiple header files with this problem. But applying the same fix to those files allowed me to compile the project.

xiaoyuantao, 2021-03-04:
Many variable declaration functions are not defined. What should I do? Is there a solution?