openmv/openmv #744: Improve Template Matching

redcrow, 2020-03-26:
### NOTE
***Forum:*** https://forums.openmv.io/viewtopic.php?f=6&t=1813&sid=cc052523870cb4995a9a134fc2bc00d4
Rotation and scale matching is really CPU intensive right now however. That said, using the log polar transforms might help optimize things.

(Rotation and scale require checking the whole image again at a different scale. Rotation the same way. With log polar it's possible to remap the image so that rotation matching is cheap).
### Steps to reproduce
1. Template matching with rotation or scale changes

### Expected behaviour
Template matching working with some degree of rotation and/or scaling involved.

### Actual behaviour
Usually fails if template and image are not lined up or scaled.

### Describe your setup

**IDE version:** 2.4.0

**Firmware version:** 3.5.2 --> OpenMV Cam H7 w/ MT9V034 Global Shutter Camera Module

**Operating system:** WIN 10 PRO

### For developers

**GCC version:**

**Build target:**

redcrow, 2020-06-06:
Any updates on this enhancement?

kwagyeman, 2020-06-06:
This is not a priority right now.

redcrow, 2020-06-06:
Understood ...Thanks for the update.

kwagyeman, 2020-06-06:
I'm working on the next release right. Doing the documentation. And then we will be working to get Edge Impulse support out for better deep learning support next. I plan to fix lower priority issues like this near the end of the year. We have a lot of higher impact objectives in the future however.

MNRobotics, 2022-12-09:
Hello,

any progress in template matching?

kwagyeman, 2022-12-09:
Nope! Way to busy.

Folks could submit PRs and contribute to the code base.