openmv/openmv #746: OV5640 driver error

kidswong999, 2020-03-30:
firmware version: 3.6.0
code:
```
import sensor, time, image

# Reset sensor
sensor.reset()
sensor.set_contrast(3)
sensor.set_gainceiling(16)
sensor.set_framesize(sensor.VGA)
sensor.set_windowing((320, 240))
sensor.set_pixformat(sensor.GRAYSCALE)
```

assert:
ValueError: Failed to set framesize!
when sensor.set_framesize(sensor.VGA)

OV5640 assert error, but OV7725 is OK. I think OV5640 driver has an issue.

[image]

kwagyeman, 2020-04-22:
Initialize in the reverse order. The issue happens because the format isn't set yet and I have strict checks. Will fix.

kwagyeman, 2020-05-12:
This is fixed.

closed 2020-05-12 by kwagyeman