openmv/openmv #749: Please add img.sum() and img.sum_of_differences() methods

bitbank2, 2020-04-07:
For motion detection and other applications where having the magnitude of the difference between two images would be helpful. In the case of motion detection, this would allow faster detection by avoiding the need to create a difference image and just provide an absolute number from two source images.

kwagyeman, 2020-05-13:
I will make get_histogram and get_stats accept a second image as an argument to compute the abs(img1-img2). Maybe also an op request for other things.

kwagyeman, 2020-05-13:
https://github.com/openmv/openmv/pull/799

I don't want to add sum/sum_of_diffs because these are not normalized values. Given we support lots of different image types and sizes its better to supply the histogram which can be used to give you whatever information you need in a normalized format that doesn't really matter to much on the image type and number of pixels.

closed 2020-05-13 by kwagyeman