openmv/openmv #751: MircoPython Graphics Library

SystemDefenseNetwork, 2020-04-11:
**Objective:**
Trying to use the LCD shield for purposes other than camera mirroring

**Research:**
[https://blog.littlevgl.com/2019-02-20/micropython-bindings](url)
[https://github.com/littlevgl/lv_binding_micropython](url)
[https://github.com/littlevgl/lvgl](url)

**Questions:**

-  Is this feasible?

**Direction**

Im willing to take on this project, I just need some sort of basic outline about how to accomplish this if it's even possible.

kwagyeman, 2020-04-11:
Hi, just alloc another frame buffer using alloc_extra_fb() and then you can use the drawing methods to draw on it.

If you want to try to implement little vgl in Python go ahead. It's going to be a tremendous amount of work but it will be very valuable.

kwagyeman, 2020-04-11:
Note we have drawing methods already. If you plan to do this you'd want to make a new MicroPython module. Wrap each library call in the library with a MicroPython handler and then add it to the modules list. Finally, you'd make all the methods operate on an image.

SystemDefenseNetwork, 2020-04-11:
Great. That should be enough to get me started.

SystemDefenseNetwork, 2020-04-14:
Doing some more digging Stm32 supports TouchGfx

https://www.touchgfx.com/product/stm32-gui/

https://www.st.com/en/embedded-software/x-cube-touchgfx.html

When I get my cameras i will build out the environment. I noticed they have prebuilt environments setup for dev boards. 

Did the OpenMV team use a Dev board or the Stm IDE suite at all? If so do you have a Peripheral Library template? 

Currently they are only supporting the following models:
STM32F0, STM32F1, STM32F2, STM32F3, STM32F4, STM32F7, STM32L0, STM32L1, STM32L4

I don't seem much for H7 devices

kwagyeman, 2020-04-14:
Um, I wouldn't put touch gfx inside of OpenMV. I'd do that the other way around. TouchGFX is like it's own OS. You should just pull what you want out of OpenMV and put that into Touch GFX.

The littleVGL could be integrated into OpenMV but not touchGFX.

SystemDefenseNetwork, 2020-04-14:
Thanks for the info! Ill stick to littleVGL since it will be better for the community.

kwagyeman, 2020-09-12:
Hi, this is going to get done now since we are making a product with groupgets with a touch screen.