openmv/openmv #764: Image.mean() corrupts memory

matsondawson, 2020-04-24:
### Steps to reproduce
1. Run this script
import image, sensor, time

clock = time.clock()
image = image.Image(256, 1, sensor.RGB565)

while(True):
    clock.tick()
    image.mean(32)
    print(clock.fps())

### Expected behaviour
Image gets averaged over 32 pixel window.

### Actual behaviour
Hardware hangs or resets.

### Describe your setup

**IDE version:** 
2.4.0

**Firmware version:** 
MicroPython v1.12-omv OpenMV v3.6.1 2020-03-30; OPENMV4-STM32H743

**Operating system:**
Windows 10 Home

kwagyeman, 2020-04-24:
It shouldn't even run.

kwagyeman, 2020-04-24:
Think you can send a PR to fix this? I'm working on an interface library right now. Code for mean is here:

https://github.com/openmv/openmv/blob/master/src/omv/img/filter.c#L134

kwagyeman, 2020-04-24:
Mmm, the issue is that the image is smaller than the kernel. The current code doesn't handle this at all right now.

matsondawson, 2020-04-24:
Ah good, shouldn't be too hard to fix.  Should I raise an error in py_mean or return if the image is too small? or both?

kwagyeman, 2020-04-24:
You couldn't that would be the easy method to prevent use of it. Yo'll have to modify a lot of things in the current method to work with an image of that size. But, I think it's doable. The inner kernel loop samples from the right pixels... but, when I go to copy it back it blows up.

kwagyeman, 2020-04-24:
Actually, looking more closely, the current code should basically handle this. There's probably a slight assumption somewhere.

matsondawson, 2020-04-24:
Is it this at filter.c:254 ?

```c
            // Copy any remaining lines from the buffer image...
            for (int y = img->h - ksize, yy = img->h; y < yy; y++) { // !! y would start at 1-32 = -31
                memcpy(IMAGE_COMPUTE_GRAYSCALE_PIXEL_ROW_PTR(img, y), // !! this would compute a pointer before the image data
                       IMAGE_COMPUTE_GRAYSCALE_PIXEL_ROW_PTR(&buf, (y % brows)),
                       IMAGE_GRAYSCALE_LINE_LEN_BYTES(img));
            }
```

kwagyeman, 2020-04-24:
Yeah, that's it. Actually, every filter method is affected by the same bug. This is the cleanup process at the end of the transfer.

I think you just need to do

for (int y = IM_MAX(img->h - ksize, 0), yy = img->h; y < yy; y++) {

kwagyeman, 2020-04-24:
Note, passing 32 for the kernel size is not a 32 wide kernel, it's a 65 pixel wide kernel. Kernel size is 2k+1.

kwagyeman, 2020-04-24:
Let me know if that works. If it does, then can you apply that fix for all places in the firmware using the same for () loop. You just need to search for

for (int y = img->h - ksize, yy = img->h; y < yy; y++) {

matsondawson, 2020-04-24:
The fix works.

kwagyeman, 2020-04-24:
K, please send a PR.

commit b9dc456b (2020-04-24): Merge pull request #766 from matsondawson/#764-Image.mean-corrupts-memory

Fix bug in filters corrupting memory if y size less than k_size.
* closes #764
closed 2020-04-24 by iabdalkader