openmv/openmv #769: OpenMV IDE not displaying the example folders contents.

funtimetinker, 2020-04-26:
### Steps to reproduce
1. Open OpenMV IDE.
2. Navigate to Help menu bar item, type search term in search box.
3. Navigate to File menu bar item, navigate to example folders.

### Expected behaviour
Examples should display folder contents.

### Actual behaviour
Folder contents are not displayed .

**IDE version:**
OpenMV IDE 2.4.0
[image]

**Operating system:**
macOS 10.15.4
### For developers

kwagyeman, 2020-04-26:
The file menu is dynamically generated when you click on it. It's not a help menu item in Mac OS. I would have to cache it to provide this feature.

As I'm not a mac user... I don't have an expectation for how this should work. What do you want to see exactly.

funtimetinker, 2020-04-26:
Everything works fine until, using the help search bar multiple times. If you make multiple searches  it will break the examples folders. You can no longer select any examples, you have to quit OpenMV IDE and re open it to be able to access the examples again.

kwagyeman, 2020-04-26:
Mmm, okay, I'll try to fix this when releasing the IDE again.