openmv/openmv #784: H7 Plus rotation_corr broken

matsondawson, 2020-05-08:
### Steps to reproduce
1. Run this code on an H7 plus.
```
import sensor, image

# Camera Init

sensor.reset()
sensor.set_pixformat(sensor.GRAYSCALE)
sensor.set_framesize(sensor.QQVGA)
sensor.skip_frames(time = 2000)

# Camera Defs
while(True):
    # Capture an image
    img = sensor.snapshot()
    img.rotation_corr(x_translation=1,y_translation=1)
```

### Expected behaviour
Image is shifted 1 pixel.

### Actual behaviour
Error is returned "MemoryError: out of fast Frame Buffer Stack memory."

### Analysis
The difference is in the OMV_UMM_BLOCK_SIZE value I believe.
It changed from 16 to 256.

I see a heap allocation using umm_init_x
```
apriltag.c:12328
void imlib_rotation_corr(image_t *img, float x_rotation, float y_rotation, float z_rotation,
                         float x_translation, float y_translation,
                         float zoom, float fov, float *corners)
{
  umm_init_x(8000); // 400 20 byte heap blocks...
  ...
}
```
Which calls this function.
I think the logic below may be incorrect.
```
umm_malloc.c:242
// Note OMV_UMM_BLOCK_SIZE is 16 on H7 and 256 on H7 plys

void umm_init_x( size_t size ) {
  // UMM_MALLOC_CFG_HEAP_SIZE = 8000
  uint32_t UMM_MALLOC_CFG_HEAP_SIZE = (size / sizeof(size_t)) * sizeof(size_t);
  // H7      if (8000 <  16*128= 2048) then fail.  (Success)
  // H7 plus if (8000 < 256*128=32768) then fail!  (*Failure*)
  if (UMM_MALLOC_CFG_HEAP_SIZE < (sizeof(umm_block) * 128)) fb_alloc_fail();
  ...
}
```

**Firmware version:**
V3.6.2 and master

### For developers

**GCC version:**
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

**Build target:**
OPENMV4P

kwagyeman, 2020-05-13:
Yeah, that was an oversight. Fixing it.

kwagyeman, 2020-05-13:
Fixed https://github.com/openmv/openmv/pull/798

closed 2020-05-13 by kwagyeman