openmv/openmv #785: SOFT_I2C_SPIN_DELAY of 128 on H7 plus prevents MLX90640 running at refresh=32

matsondawson, 2020-05-08:
The configured spin delay in omv_boardconfig.h on the H7 plus is 128, and on the H7 is 32.

For some reason when fir is inited with refresh=32 with a SOFT_I2C_SPIN_DELAY of 128 the MLX90640 driver throws the error "OSError: Failed to read the MLX90640 sensor data!", if I set it to 32 it works again.

What is the value calculated from?

iabdalkader, 2020-05-08:
It was increased to fix the AMG8833 timing. It seems we have to switch to the HW I2C to make everything work on all cams.

matsondawson, 2020-05-08:
This is probably related to the refresh rate of 64 Hz not working on the H7 for the same reason.

iabdalkader, 2020-05-09:
> This is probably related to the refresh rate of 64 Hz not working on the H7 for the same reason.

The delay value is not calculated it's fixed, it should be but it's not. I can use the HW I2C for the AMG but the MLX sensors requires controlling the stop, start and repeated start conditions which the ST HAL doesn't support. I'm not sure how to fix this yet. I could use the HW I2C for the AMG sensor and keep the soft i2c with the old delay for MLX sensors.

iabdalkader, 2020-05-10:
I got the MLX90640 working with the HW I2C too, so only the old MLX will use the soft-i2c. Also got 64Hz mode working with I2C set to the fastest baudrate, but also had to increase the retries in MLX_API frame_read.

iabdalkader, 2020-05-10:
I didn't test all the sensors on all cams, will do that later, but all of the 3 sensors that we support are now working with the H7.

matsondawson, 2020-05-11:
Nice work, awesome!

iabdalkader, 2020-05-11:
Fixed.

closed 2020-05-11 by iabdalkader