openmv/openmv #806: HD plus has garbage on screen in HD and QHD with JPEG pixformat

matsondawson, 2020-05-21:
### Steps to reproduce
1. Run standard helloyworld.py, with these settings
sensor.set_pixformat(sensor.JPEG)
sensor.set_framesize(sensor.HD)
[image]

### Expected behaviour
Standard camera image rendered.

### Actual behaviour
Has camera image with some garbage pixels.

### Describe your setup

**IDE version:**
2.4.0

**Firmware version:**
Latest master build from git.

**Operating system:**
Windows 10

### For developers

**GCC version:**
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0

**Build target:**
H7 Plus

iabdalkader, 2020-05-21:
Can you test again after doing a git pull ? I've just merged 2 PRs with JPEG and OV5640 fixes.

kwagyeman, 2020-05-21:
This has been fixed with the latest jpeg code fix.

matsondawson, 2020-05-21:
Awesome. Closing.

closed 2020-05-21 by matsondawson