openmv/openmv #813: vflip ov7725 with V3.6.4 firm on M7

pyresChris66, 2020-05-27:
### Steps to reproduce
1. sensor.vflip(True) with ov7725

### Expected behaviour
vertical rotated image

### Actual behaviour
vertical rotated image with different colors

### Describe your setup
**IDE version:**
V2.4.0
**Firmware version:**
V3.6.4
**Operating system:**
windows 10

### For developers
**GCC version:**
7.5.0
**Build target:**
OPENMV3

kwagyeman, 2020-05-27:
What imaging mode are you using?

pyresChris66, 2020-05-27:
RGB565, VGA

kwagyeman, 2020-05-27:
Ah, yeah, then it's bayer. The issue is that the readout window shifted by 1 pixel messing up the bayer image data.

RGB565, VGA doesn't fit on the H7 regular so we silently convert to BAYER.

kwagyeman, 2020-05-27:
Will fix.

kwagyeman, 2020-05-30:
Fixed: https://github.com/openmv/openmv/pull/819

closed 2020-05-30 by kwagyeman