openmv/openmv #817: Can't Reproduce the Remote control

Owen718, 2020-05-28:
### Steps to reproduce
**The jpeg streaming is successfull,but when I want to reproduce the popular_features_as_the_remote_device.py ,data results are not returned.**
like image:
[https://s1.ax1x.com/2020/05/28/te7HZ6.png](url)

Please help me,thank you!
### Describe your setup

**IDE version:2.4.0**

**Firmware version:3.6.2**

**Operating system:windows10**

kwagyeman, 2020-05-28:
You need a second OpenMV Cam.

kwagyeman, 2020-05-28:
You're going to have to give some more info on what you are doing.

kwagyeman, 2020-05-28:
Also, only the H7 was tested. It's unknown if the M4/M7 work.

Owen718, 2020-05-28:
My openmv is openmv H7 ,ov7725.
I am testing the exe_apriltag_detection,I want to get the tags data.

Owen718, 2020-05-28:
I want to get the jpeg steaming and a RGB stastics data at the same time. 
Thank you very much!

kwagyeman, 2020-05-28:
Okay, and you are running the script to send data to a PC over pyserial?

Owen718, 2020-05-28:
Yeah,of course.

kwagyeman, 2020-05-28:
I need to know what's wrong. That's code is extensively tested. Please tell me exactly what you think is going wrong. I'm not restesting it without you giving me a hint. E.g. what is your setup, what is the master device, what is the slave, what comm channel are you using.

Owen718, 2020-05-28:
I can't get returned data in the  popular_features_as_the_remote_device.py . The master device is my computer,the slave device is my  openmv4 H7.I did not change the code.It's very confusing.

Owen718, 2020-05-28:
The run image :
https://imgchr.com/i/teOp9S
https://imgchr.com/i/teO91g

kwagyeman, 2020-05-28:
Okay, I understand what mode you are using. What does the script print out when you try to get the JPG image? The script just saves the JPG image to disk. It doesn't stream it.

Owen718, 2020-05-28:
I know that,but I can not get the JPG image in disk.And can't get the apriltag data also....Data other than the image stream is not available to get.

kwagyeman, 2020-05-28:
Oh, are you running the script with the IDE attached? You cannot do that. You have to save the script to the OpenMV Cam and then reset the OpenMV Cam. Verify the script has been saved when it reconnects to the PC again, and then you should be able to use the USB connection.

kwagyeman, 2020-05-28:
You can verify the script has been saved by opening main.py and seeing the file contains the script from the IDE.

Note that you cannot use the IDE in the process once the script is onboard.

Owen718, 2020-05-28:
The RPC module can't use wih the IDE. Of course I do like that.

Owen718, 2020-05-28:
I do that again , like you said. But still NO data returned....

kwagyeman, 2020-05-28:
Is the script saves to the camera successfully?

It works for me over here... need some info on debugging.

Owen718, 2020-05-28:
The script saves to the camera successfully...It's very confusing,it let me input com name ,but did not return any data after that.What info do you need?

kwagyeman, 2020-05-28:
It kinda should just work after that.

I'm guessing this returns None:

https://github.com/openmv/openmv/blob/master/tools/rpc/rpc_popular_features_as_the_controller_device_example.py#L110

...

The timeouts are probably too tight. But, the interface library automatically adjusts timeouts to make syncing work.

Put this in the script library: https://github.com/openmv/openmv/blob/master/tools/rpc/rpc.py#L289

Add a print(binascii.hexlify(result)) before the return.

And add another print in the put data:

https://github.com/openmv/openmv/blob/master/tools/rpc/rpc.py#L292

...

I spent enough time debugging this code to read the binary data commands back and forth and I should be able to tell you what's wrong based on what traffic is moving back and forth.

Owen718, 2020-05-28:
I think I did not understand how to change the rpc.py.
--------------------------------
COM9 : OpenMV Cam USB COM Port (COM9) [USB VID:PID=1209:ABD1 SER=6 LOCATION=1-2:x.0]

Please enter a port name: COM9

Traceback (most recent call last):
  File "c:\Users\Owen\Desktop\rpc\rpc_popular_features_as_the_controller_device_example.py", line 132, in <module>
    exe_apriltag_detection()
  File "c:\Users\Owen\Desktop\rpc\rpc_popular_features_as_the_controller_device_example.py", line 66, in exe_apriltag_detection
    result = interface.call("apriltag_detection")
  File "c:\Users\Owen\Desktop\rpc\rpc.py", line 180, in call
    return self.__get_result(recv_timeout) if self.__put_command(self._hash(name, len(name)), data, send_timeout) else None
  File "c:\Users\Owen\Desktop\rpc\rpc.py", line 148, in __put_command
    self.put_bytes(out_header, self._put_short_timeout)
  File "c:\Users\Owen\Desktop\rpc\rpc.py", line 292, in put_bytes
    print(binascii.hexlify(result))
NameError: name 'result' is not defined

kwagyeman, 2020-05-28:
put_bytes() doesn't have a result variable... it takes `data`. So, pass `data`.

Owen718, 2020-05-28:
I`COM9 : OpenMV Cam USB COM Port (COM9) [USB VID:PID=1209:ABD1 SER=6 LOCATION=1-2:x.0]

Please enter a port name: COM9

b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
[... 622 lines cut ...]
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'
b'00000000'
b'0912617f4184000000002b11'`

Owen718, 2020-05-28:
`while(True):
    #exe_face_detection() # Face should be about 2ft away.
    #exe_jpeg_RGB_stastics()# exe_person_detection()
    # exe_qrcode_detection() # Place the QRCode about 2ft away.
    # exe_all_qrcode_detection() # Place the QRCode about 2ft away.
    exe_apriltag_detection()
    exe_datamatrix_detection()
 
    #exe_all_apriltag_detection()
    # Place the Datamatrix about 2ft away.
    # exe_all_datamatrix_detection() # Place the Datamatrix about 2ft away.
    # exe_barcode_detection() # Place the Barcode about 2ft away.
    # exe_all_barcode_detection() # Place the Barcode about 2ft away.
    # exe_color_detection()
    #exe_jpeg_snapshot()`

### in the rpc_popular_features_as_the controller_device_example.py.

Owen718, 2020-05-28:
**I command exe_apriltag_detection:**
Please enter a port name: COM9

b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'
b'09127e662afc00000000fb6a'
b'00000000'

kwagyeman, 2020-05-28:
The OpenMV Cam is not responding at all. From the trace, the PC is issuing the command to do something and the camera never responds to that. There's not even an attempt from the camera to do anything.

Please disconnect the camera from your PC and then plug it back in and verify main.py on the camera is the example script with an external editor like notepad++ or etc.

Owen718, 2020-05-28:
I really do like you say...I copy the popular_features_as_the_remote_device.py and change it's name .And disconnect the camera from PC and plug it back to run it...But the print is still wrong.

kwagyeman, 2020-05-28:
Did you save use safety remove eject? Please verify the script is actually updated on the camera and that the USB VCP interface is uncommented as the only interface.

Owen718, 2020-05-28:
The script is actually updated on the camera,I have used safety remove eject.The USB VCP interface is uncommented.I didn't change it.

kwagyeman, 2020-05-28:
This is uncommented:

https://github.com/openmv/openmv/blob/master/scripts/examples/34-Remote-Control/popular_features_as_the_remote_device.py#L65

And this is commented:

https://github.com/openmv/openmv/blob/master/scripts/examples/34-Remote-Control/popular_features_as_the_remote_device.py#L61

? The UART interface is not the same as the VCP one. The timeouts on it are very tight. Much tighter than the VCP interface.

kwagyeman, 2020-05-28:
Not really sure what to do... the PC side is not reading any activity from the OpenMV Cam.

Can you add a this here:

https://github.com/openmv/openmv/blob/master/scripts/examples/34-Remote-Control/popular_features_as_the_remote_device.py#L268

```
def loop_cb:
    print("FOOBAR00")
interface.setup_loop_callback(loop_cb)
```

Then open the USB VCP port with a serial terminal program and very that's printing...

Owen718, 2020-05-28:
I try like you said,add the code in L268,but I did not receive any bytes.

Owen718, 2020-05-28:
It's 00:45am right now.I am tired,have to go sleep.Thank you again!I will continue find the reason....

kwagyeman, 2020-05-28:
I don't know what's wrong then. The script on the PC is definitely not getting any data at all from the camera and you are telling me the script on the camera is not printing any alive status information from a script that it should be running. So, there's an issue with the script running on the camera. I will verify all this works in a bit.

kwagyeman, 2020-05-28:
Yeah, I'll verify this is not broken.

kwagyeman, 2020-05-30:
I can confirm the library works with the latest firmware, steps:

1. Open popular_features_as_the_remote_device.
2. Comment: interface = rpc.rpc_uart_slave(baudrate=115200)
3. Uncomment: interface = rpc.rpc_usb_vcp_slave()
4. Tools->Save Script to OpenMV Cam.
5. Tools->Reset OpenMV Cam.
6. Verified the file was saved as main.py on the camera after it remounted.
7. python -u rpc_popular_features_as_the_controller_device_example.py
8. Typed in the COM port number.
9. It detected by face and prints the location.

closed 2020-05-30 by kwagyeman