openmv/openmv #825: OpenMV is just crashing

bart, 2020-06-07:
I got my OpenMV H7 today. Version is 10/27/18. I'm on the latest firmware updated via IDE.
It currently looks like it has a hardware issue which is super annoying!!!

Simple scripts like blinking the LED works. But as soon as it gets a bit more complicated like detecting Apriltags (from the examples) the script crashes after one or two seconds. Not sure if it is a power supply issue (running on USB) or anything else. Cannot use the OpenMV module which really sucks!

Any ideas? Thanks!

kwagyeman, 2020-06-07:
Let me debug.

kwagyeman, 2020-06-07:
Hi, I just loaded the latest firmware on an H7 regular with an OV7725 camera. I ran a test using the find_apriltags script and it worked fine.

I literally just released new firmware about 30 minutes ago. Can you restart openmv ide? It should download and install this firmware. Next, please run the hello world script, confirm that works, then run an apriltags script and let me know which one it is.

bart, 2020-06-08:
Hey @kwagyeman thanks for you fast response! I am on the same camera, OV7725 with firmware 3.6.4. The hello world example runs without any crashes. Face detection examples are working but crashing after a while (~2 mins), too. As soon as I start the find_apriltags.py example script it crashes immediately. Is there any chance to get the reason like exception logging feature so that I can send you additional information?

I'm working on a Macbook Pro, so it's OSX. I can try it on Windows, too but I think it's more a device issue than an OS problem.

bart, 2020-06-08:
Btw. same issue occurs when I run "Find Circles Example" for example. It's not just the Apriltags example. Maybe it's a hardware or firmware issue. Hope we can find the reason soon, so that I can send the device back if it is broken.

bart, 2020-06-11:
Thanks for your response! I got it from Amazon: https://www.amazon.de/gp/product/B07TJY7N75
For me it looks like an original OpenMV board - what do you think? Version is 10/27/18.
Should I order another one and send the first one back?USB connector looks good.

bart, 2020-06-11:
I decided to buy a new H7 Plus from your partner Antratek and will send back the broken one. Hope this will solve the issue.

closed 2020-06-11 by bart

kwagyeman, 2020-06-11:
I don't know how to debug the issue. I'm guessing it's broken since I can't reproduce the error.

bart, 2020-06-11:
Yeah I know, it's super difficult. That's why I expected a hardware issue from the beginning. Will let you know if the new one works.

bart, 2020-06-15:
Really looks like it was a hardware issue. Got my OpenMV Cam H7 Plus today and it's working now. Thanks for this amazing project!