openmv/openmv #840: Updating firmware to include newer version of ulab

mz3-andrey, 2020-06-24:
Hey guys,
The openmv firmware's version of ulab (0.26) has a bug with inconsistent slicing (https://github.com/v923z/micropython-ulab/issues/128).
@v923z mentioned that @iabdalkader could now easily include the newest version of ulab as he modified his code from what I understand.
I could try to figure out how to compile the firmware with the new version of ulab, but was hoping it could also be done officially to also help other users too.
Thank you!

iabdalkader, 2020-06-24:
I did update ulab to 0.50.2, which should include the fix for this issue, right ? It will be in the next release. If you want the fix now you could build the firmware from source.

mz3-andrey, 2020-06-25:
Great it works! Thanks

closed 2020-06-25 by mz3-andrey