openmv/openmv #842: compile error

BenjaminSavioz, 2020-06-29:
I try to compile the ../openmv-master/src/micropython/mpy-cross on ubuntu 20.04 and i have this error:

python3 ../py/makeversionhdr.py build/genhdr/mpversion.h
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 94, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 67, in make_version_header
    omv_git_tag, omv_git_hash = get_version_info_from_git("../../../")
TypeError: cannot unpack non-iterable NoneType object
make: *** [../py/py.mk:233: build/genhdr/mpversion.h] Error 1

please can you help me

iabdalkader, 2020-06-29:
Did you clone the repo or download the sources ? It looks like it's not a git repo.

BenjaminSavioz, 2020-06-30:
I found my error, I cloned on my host OS and I made a file to share between my VM and my host. But now I clone the code in the VM and it works.
but i just have a question for the firmware instalation. if i have a PC on windows and i install the vm on it, i need to follow the guide for the linux install too?

Thank's for the answer.

kwagyeman, 2020-06-30:
Yes

closed 2020-06-30 by kwagyeman