openmv/openmv #852: Do OpenMV camera can obstacle avoidance and find way like Lidar or sonic sensor? any camera do that?

shoaibsayed1, 2020-07-16:
Do OpenMV camera can obstacle avoidance and find way like Lidar or sonic sensor? does any camera can do that?
my graduation project is about obstacle detection and find way to search fire and identify different object. I have sonic sensor and professor want something image base which can detect any object and situation and find his own way to travel more.
I have raspberry pi 4 model b.

kwagyeman, 2020-07-16:
We can do image classification. Using this you can avoid if you see a particular type of class ahead.

-- Kwabena

shoaibsayed1, 2020-07-16:
> We can do image classification. Using this you can avoid if you see a particular type of class ahead.
> […](#)

any kind of reference ? any suggestion ? I am too much tensed. is their any better option to avoid obstacle and find the path way with image based system? in YouTube their is nothing with openmv+rasepberry pi. alls are alone image detection only by OpenMV.

kwagyeman, 2020-07-16:
Hi, we are about to release an interface library that makes it easy to connect the camera to a Pi. However, you are a little early for both the features discussed.

That's said, cameras don't do object avoidance in general since they don't really give distance.

-- Kwabena

shoaibsayed1, 2020-07-16:
> Hi, we are about to release an interface library that makes it easy to connect the camera to a Pi. However, you are a little early for both the features discussed. That's said, cameras don't do object avoidance in general since they don't really give distance.
> […](#)

I have 5 month to complete my project I wish I can be the first one who's gonna test feature and do feedback from the international clouds.
As i saw in YouTube OpenMV detect line, color and follow them, if it possible while they follow different object. they can avoid also different obstacle and find something as a task. through this following feature will work as searching features.

iabdalkader, 2020-07-16:
@shoaibsayed1 Hello, please post any questions you have on the forums https://forums.openmv.io/ and not on github, closing this issue now.

closed 2020-07-16 by iabdalkader